shariltumin/esp32-cam-micropython-2022 #2: lwip.setsockopt() option not implemented

zh-nj, 2022-09-23:
Warning: lwip.setsockopt() option not implemented,esp32 1.19firmware works.

shariltumin, 2022-09-23:
I apologize for any inconvenience this has caused you. 

When I attempted to run the official firmware, I discovered that lwip was missing. I'm not sure what firmware you're running.

MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
>>> 
>>> lwip.setsockopt()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'lwip' isn't defined
>>> import lwip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'lwip'
>>> 

micropython$ git describe 
v1.19.1-443-g30e50ab19

micropython/ports/esp32$ egrep LWIP *
modsocket.c:// Note: This port doesn't define MICROPY_PY_USOCKET or MICROPY_PY_LWIP so

LWIP appears to be disabled in standard DISTIBUTION.

micropython/ports/esp32$ diff mpconfigport.h mpconfigport.h-ORIG 
132d131
< #define MICROPY_PY_LWIP			    (1)

export IDF_PATH="$HOME/esp/esp-idf-440"
source $IDF_PATH/export.sh
make V=1 BOARD=GENERIC

Attempting to compile mp version v1.19.1-443-g30e50ab19 to include lwip by including MICROPY PY LWIP in mpconfigport.h results in an error for esp-idf 440, 441, and 442.

shariltumin, 2022-09-23:
If you receive the following error message: Warning: lwip.setsockopt() option not implemented, this is because MICROPY_PY_WEBREPL  and  MICROPY_PY_USOCKET_EVENTS are not enabled in the firmware.

closed 2022-09-23 by shariltumin