shariltumin/esp32-cam-micropython-2022 #3: Functions to flip the image are necessary

FerriGit, 2022-11-08:
When you take pictures and save them directly into the SDcard, you want the image saved in the right position. So, sometimes it is necessary to flip ithe picture. Functions like camera.vflip(0 or 1) and camera.hmiror(0 or 1) are used to get this done.
I see you have implemented the function camera.flip(), but the horizontal mirror function is also necessary.
Could you implement the missing function?
Thank you,
Fernando Echeverri
ferri_1@yahoo.com

shariltumin, 2022-11-17:
I will see what I can do. The next firmware update will incorporate hmirror functionality.

FerriGit, 2022-11-19:
Thank you very much for your attention and your contributions.

Sincerely,

Fernando Echeverri

***@***.***

shariltumin, 2022-11-21:
Please try firmware-20221121/firmware.bin

FerriGit, 2022-11-28:
Hello, Mr. Sharil

I regret to inform you that this firmware did not work at all. It keeps rebooting, as you can see below:

 

Rebooting...

ets Jun  8 2016 00:22:57

 

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:4396

ho 0 tail 12 room 4

load:0x40078000,len:13884

load:0x40080400,len:3340

entry 0x40080618

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

 

Core  0 register dump:

PC      : 0x40180a89  PS      : 0x00060d30  A0      : 0x8017da58  A1      : 0x3ffcd410

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb7f08  A5      : 0x00000000

A6      : 0x3ffbb448  A7      : 0x3ffbb3fc  A8      : 0x00000000  A9      : 0x00002000

A10     : 0x00000000  A11     : 0x00000080  A12     : 0x00000000  A13     : 0x00000001

A14     : 0x00060b23  A15     : 0x00060b20  SAR     : 0x00000020  EXCCAUSE: 0x0000001c

EXCVADDR: 0x0000006c  LBEG    : 0x4008cdf4  LEND    : 0x4008ce10  LCOUNT  : 0xffffffff

 

Backtrace:0x40180a86:0x3ffcd4100x4017da55:0x3ffcd430 0x401664bc:0x3ffcd450 0x401667b7:0x3ffcd470 0x4015b989:0x3ffcd490 0x401593cc:0x3ffcd4d0 0x4019e345:0x3ffcd4f0

ELF file SHA256: 6af690dc11401755

 

It keeps rebooting. I tried it with two different ESP32-CAM, and I got the same result.

 

Best regards,

Fernando

shariltumin, 2022-11-28:
I downloaded the firmware from github. Then erase and re-flash firmware.

This is what I get right after flashing the firmware. 1st. time power btn reset:

Power on rest (push reset btn):

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4396
ho 0 tail 12 room 4
load:0x40078000,len:13884
load:0x40080400,len:3340
entry 0x40080618
Performing initial setup
MicroPython v1.19.1-705-gac5934c96-kaki5 on 2022-11-21; ESP32 CAMERA module (KAKI5) with ESP32
>>>

And then doing the machine.reset():
machine.reset():
>>> import machine
>>> machine.reset()
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4396
ho 0 tail 12 room 4
load:0x40078000,len:13884
load:0x40080400,len:3340
entry 0x40080618
MicroPython v1.19.1-705-gac5934c96-kaki5 on 2022-11-21; ESP32 CAMERA module (KAKI5) with ESP32
>>>

As I can see, the boot sequence were the same as your up to "entry 0x40080618". Instead of getting REPL you get
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Strange that it is on Core 0. Micropython and camare module run on Core 1. Wifi and Bluetooth run on Core 0.

I suggest you erase the flash memory and re-flash the firmware. 

Disconnect GIOP00 and GND. Connect a terminal to the serial port where you have your board connect. Push the RST button

You should get: 
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
...
...
Performing initial setup
and REPL
if everything works.

FerriGit, 2022-12-01:
Dear Mr. Sharil. Based on your comments, I tracked my program, the one that works well with the first firmware, and I found out that the problem was caused by the following instruction (my the ESp32-CAM is an access point):

ap.config(essid= _ssid, password= _key, channel=11)

It is channel=11 that produced the meditation error, so I changed the sentence to: ap.config(essid= _ssid, password= _key) and the program runs well. Using another channel does not fix the problem. I do not understand what happens, but it seems related to the new Micropython version 1.19.

 

Btw: the mirror function is working well.

Thank you very much for your collaboration.

Best regards,

Fernando

closed 2023-05-21 by shariltumin