shariltumin/esp32-cam-micropython-2022 #7: Partial pictures.

davefes, 2022-11-17:
About 25% of the time I take a picture is it missing part of the image.  It seems to be the missing EOF, which is causing my server to reject it.

I have found that any delay between camera.init() and doing any of the settings can provoke similar issues.

What is the correct way to init the camera and apply wanted settings?

I also have found that I have to do a POWER_ON reset or I can get failed camera init as well.

Would really appreciate some feedback on these issues.

shariltumin, 2022-11-17:
Before making any other changes, run camera.init() first.

If camera.init() fails, the board must be hard reset.

POWER ON A push button can be used for this, or in MicroPython, do:

import machine
machine.reset()

Image quality and stability are dependent on the following factors:
1) Consistent power supply
2) The OV2640 sensor's quality
3) The connection at the flex 24 connectors
4) The MCU

I'd start by checking the power supply.

Use 5V from the USB-to-Serial to power the ESP32-CAM, i.e. connect 5V from the USB-to-Serial to 5V on the ESP32-CAM board. Make use of a USB hub with an external power supply. PC USB ports may not have sufficient current.

Change to another OV2640 sensor, i.e., the lens.

Run this stress check.

import camera
camera.init()
while True:
    img=camera.capture()
    print(len(img))

You should get reasonably consistent numbers output.

davefes, 2022-11-17:
Thank you for your comments.

As per https://github.com/orgs/micropython/discussions/9345 I found that a POWERON_RESET does a better job.

After another late night, specifically changing camera.framesize and camera.quality, I seem to get fewer corrupted pictures using 14 (FRAME_UVGA) and quality 30.

During testing I am powering the ESP32-CAM via a FTDI programmer, but with an added 1000uF cap on the 5V.  I am using the flash LED so that could be putting a heavy-load on the AMS1117.  I check for supply "dips" using a DSO.  In final use it will be powered off a Li-ion cell.

To handle corrupt files I comment-out the print(len(img)) statement and do the following:
```py
#  save image
with open('image.jpeg', 'wb') as outfile:
    outfile.write(img)

f = open('image.jpeg', 'rb')
first2bytes = f.read(2)
f.close()

valid_jpeg = b'\xff\xd8'

if (first2bytes == valid_jpeg):
    print ('success')
    gc.collect()
    import send_file
else:
    print ('not a valid jpeg file')
    utime.sleep(5)
    machine.reset() #  change to a power-on reset implementation
```
I will consider the other suggestions you have made.

shariltumin, 2022-11-17:
Thanks, your POWERON_RESET can be very useful to get machine.PWRON_RESET. machine.reset() gives machine.HARD_RESET only.

davefes, 2022-11-17:
The previous test for a valid_jpeg file only appears to sort-out the files that do not have the correct header, which happens occasionally.  Files that can be missing some on the bottom of the picture still have the correct header!

Thanks to Google it seems you can check for a proper jpeg file ending ... in "all or most cases".

```py
#  save image
with open('image.jpeg', 'wb') as outfile:
    outfile.write(img)

#  valid jpeg ending
valid_jpeg = b'\xff\xd9'

#  now check for the correct ending
with open('image.jpeg', 'rb') as file:
    try:
        file.seek(-2, 2)
        while file.read(1) != b'\n':
            file.seek(-2, 1)
    except OSError:
        file.seek(0)

    last_line = file.readline()

last_chars = last_line[-2:]
print('Last 2 characters : ', last_chars)

if (last_chars == valid_jpeg):
    print ('success')
    gc.collect()

    import send_file
else:
    print ('not a valid jpeg file')
    utime.sleep(5)
    config.poweron_reset.on()
```

davefes, 2022-11-17:
I suspect even checking for the correct ending that one might still get a "cropped" image.  So far, at least, the server seems to handle those "cropped" images properly.

Stress testing now begins.

davefes, 2022-11-19:
After many more tests I can see the image that I am saving on flash is the same as the image on my server, with no cropping.

Appears that disconnecting the WiFi immediately after soc.close() could be the cause of the "cropping".  I have seen "timing-related" issues before  with WiFi on the ESP32.  A 1 second delay after sock.close(() seems to fix that problem.

Now, I just need to work out why the server hangs waiting for data, even after getting a complete image.

closed 2022-11-19 by davefes