shariltumin/esp32-cam-micropython-2022 #8: Are the timers in the firmware?

davefes, 2022-11-20:
I am trying to implement a SW watchdog timer that will allow me to do a poweron_reset.  When I import the watchdog timer methods into my main program I get a guru meditation error.
```py
wdt_timer = machine.Timer(3) #  hardware timer for ESP32
wdt_timer.init(period=1000, mode=machine.Timer.PERIODIC, callback=lambda t:wdt_callback())
```

shariltumin, 2022-11-21:
I have the same problem too. I've now compiled a new firmware. Please try firmware-20221121/firmware.bin

davefes, 2022-11-21:
I will after I sort out a much bigger issue.  Every time I mounted the camera outside on a post and triggered the camera, the flash would stay on much longer sometimes 5-10 seconds.  After these long exposure events the camera locks-up.

I put some tape over the camera lens and that problem goes away.

#  camera settings
camera.pixformat(0) #  0:JPEG, 1:Grayscale (2bytes/pixel), 2:RGB565
camera.framesize(14) #  1:96x96, 2:160x120, 3:176x144, 4:240x176, 5:240x240
                    #  6:320x240, 7:400x296, 8:480x320, 9:640x480, 10:800x600
                    #  11:1024x768, 12:1280x720, 13:1280x1024, 14:1600x1200
                    #  15:1920x1080, 16:720x1280, 17:864x1536, 18:2048x1536
camera.quality(20) #  [0,63] lower number means higher quality
camera.contrast(0) #  [-2,2] higher number higher contrast
camera.saturation(0) #  [-2,2] higher number higher saturation. -2 grayscale
camera.brightness(0) #  [-2,2] higher number higher brightness. 2 brightest
camera.speffect(2) #  0:,no effect 1:negative, 2:black and white, 3:reddish,
                   #  4:greenish, 5:blue, 6:retro
camera.whitebalance(0) #  0:default, 1:sunny, 2:cloudy, 3:office, 4:home
camera.aelevels(0) #  [-2,2] AE Level: Automatic exposure
camera.aecvalue(0) #  [0,1200] AEC Value: Automatic exposure control
camera.agcgain(0)  #  [0,30] AGC Gain: Automatic Gain Control

It seems that brightness or picture detail maybe determines how long it takes to finish the camera.capture()

davefes, 2022-11-21:
New firmware with the above camera settings:
2 pictures with green bands at the top (10-20%) and 1 picture 100% OK

Changed camera.framesize to 10:
2 pictures with a darker contrast band at the top (10-20%).

Will leave the watchdog timer tests for later.

davefes, 2022-11-21:
Changed camera.framesize back to 14 and everything is now OK.

What it looks like to me is the settings were changing during picture capture.  I previously mentioned that any delays between camera.init() and doing the settings seemed to mess things-up.  I will re-check that and try a delay after the camera settings.

Would it help to leave out any default settings and only do the ones you want to change?  And they are?

The long flash pulses seem to happen when you are in a bright environment.  Does it take longer to do the capture when it is sunny?

shariltumin, 2022-11-21:
Are you trying to save images to SDCard? Writing to SDCard will trigger the flash.

davefes, 2022-11-21:
No.  I save the image to flash and then send it over WiFi.

I can also get very long flash pules with my finger blocking out the light and then the unit "hangs" until my power_reset()

davefes, 2022-11-21:
I started this journey here:
https://randomnerdtutorials.com/esp32-cam-pir-motion-detector-photo-capture/
so I am aware about the SD card issue.

davefes, 2022-11-21:
I should have said, I save the image to flash then fire-up the WiFi, send the image and then disconnect.

shariltumin, 2022-11-21:
May I ask why you need to save the captured image to a file before you send it. Do you use TCP or UDP to send the image?

shariltumin, 2022-11-21:
I assume you have a TCP server somewhere and you are sending image from esp32-cam to that server. That server is serving as a HUB so that you can send image to whatever services, eg. Gmail.  How often you are taking photos?

davefes, 2022-11-21:
Yes.  Eventually, that is the plan.  I send daily logs from our off-grid house and the next step is to send pictures.  During testing I can repeatedly take pictures with about a 10 second delay for the server to start listening again.

Early-days in the current stress-test but deinit() the camera after turning the flash off has been error-free.  Suggests doing the camera.deinit() after sending the image is not the right thing to do.

Again there seem to be some "timing" constraints recgarding camera.init, making the settings and camera.deinit().

shariltumin, 2022-11-21:
I have those problem too, with bad sensor. I suggest you try with another if you have a spare one. esp32-cam and OV2640 are cheap, we can not expect mush quality out of them. Try smaller framesize.

davefes, 2022-11-21:
Two more just came in, so will try another camera.  camera.framesize(10) always shows a darker band at the top.

shariltumin, 2022-11-21:
I am streaming live with framesize 10 i.e 800x600 now. I don't see darker band anywhere. I see a lot of noise.

davefes, 2022-11-21:
10 good snaps in a row.  I would happily pay 10 times more for a good camera ... if that is the problem.

Stress-testing at framesize 14 and quality 10 now begins.  Will report back in an hour or so.

davefes, 2022-11-21:
20-30 more pictures one looks over-exposed and the rest are very poor quality (poor resolution).  Found this comment:
> 
I have dug deeper…
You have to be very careful what combination of parameters you use and in what order you use them.
I had more success by commenting out the parameters not needed.
Extra care has to be taken when any of the Auto controls are used, as subsequent parameters are ignored.
at:
https://randomnerdtutorials.com/esp32-cam-ov2640-camera-settings/

Does seem a bit "hit and miss".

davefes, 2022-11-22:
Well, tested 3 cameras, each with image distortions.

Thought I would go back to the original Arduino sketch that got me going and I get lovely coloured pictures.  Didn't want to tackle Arduino S/W as all my other stuff is using Micropython, O well.

Thanks for trying.  I am happy to test further images.  Perhaps if you leave a note on this thread I will see it.

closed 2022-11-22 by davefes