shariltumin/esp32-cam-micropython-2022 #10: USSL support

davefes, 2022-12-02:
I tried adding in ussl.py and get the following error:
```py
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
  File "ussl.py", line 36, in wrap_socket
RuntimeError: maximum recursion depth exceeded
```
The owner of umail suggesting "flattening the recursion", which I will try to do as a last ditch attempt.

Would it be a big job putting ussl in the last image you produced or will it exceed some memory requirement?

Thanks,
Dave

shariltumin, 2022-12-03:
For SSL, please try firmware-20221203.

davefes, 2022-12-04:
Using this image, uMail https://github.com/shawwwn/uMail and an example of sending the image https://github.com/shawwwn/uMail/issues/2 it worked ... first time!

Thanks for your effort again.

davefes, 2022-12-04:
Finished

closed 2022-12-04 by davefes