shariltumin/esp32-cam-micropython-2022 #14: Better firmware

andim27, 2023-02-06:
Could you make firmware with BLE support?

askoriy, 2023-05-02:
yes, it would be good feature. I assume that firmware doesn't fit to standard 4MB flash, but its possible to solder larger spi flash chip

shariltumin, 2023-05-03:
I don't believe esp32_cam with BLE or Bluetooth Classic will be a viable system. When using SPP with the ESP32 BLE and Bluetooth Classic, the maximum MTU is 23 bytes and 990 bytes, respectively. To send an image file over SPP, it must be sent in chunks. These chunks must then be reassembled by the receiver.

I'm curious if we have BLE clients that can effectively handle data stream chunks. Sending 2K Bytes of framebuffer over BLE to such a device is one way to test. This can be tested with the official esp32 firmware.

By removing network support, it is possible to incorporate BLE into the esp32_cam firmware. I've actually done it before. I also have a Bluetooth Classic test device that supports SPP. https://github.com/shariltumin/esp32-bluetooth-classic-micropython

Chunking and reassembling may be possible only for still images.

shariltumin, 2023-05-21:
For those who need firmware with BLE please have a look at [firmwares-20230521](https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/firmwares-20230521)