shariltumin/esp32-cam-micropython-2022 #15: Invalid Header

deus-est-machina, 2023-02-27:
Hi :)

Thanks for the great work and the proper README instructions.

Unfortunately, I am receiving something weird back after I reset the module:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
ets Jul 29 2019 12:21:46
```

Any Ideas?

deus-est-machina, 2023-02-28:
I am using `Thonny,` yes. I have the Erase flash checked. 

```
...\Local\Programs\Thonny\python.exe -u -m esptool --port COM6 erase_flash
esptool.py v4.4
Serial port COM6
Connecting...........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 42.28MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 44:17:93:7b:fd:dc
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 11.5s
Hard resetting via RTS pin...
Erasing done
------------------------------------

...\Local\Programs\Thonny\python.exe -u -m esptool --port COM6 write_flash --flash_mode keep --flash_size detect 0x1000 "C:/Users/ruben/_work/projects/esp32/MicroPython v1.19.1-705-gac5934c96-kaki5-2022-12-03ESP32CAMERAwSSL.bin"
esptool.py v4.4
Serial port COM6
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 42.28MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 44:17:93:7b:fd:dc
Stub is already running. No upload is necessary.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00033fff...
Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings.
Compressed 208112 bytes to 36251...
Writing at 0x00001000... (33 %)
Writing at 0x00017750... (66 %)
Writing at 0x0002fa4b... (100 %)
Wrote 208112 bytes (36251 compressed) at 0x00001000 in 3.9 seconds (effective 427.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done!
```

shariltumin, 2023-02-28:
Hi, Thanks for the info. I am not sure if you were flashing the correct firmware. The firmware size is 1270928 

flash log on my part gave  these numbers:
...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00137fff...
Compressed 1270928 bytes to 821711...
Wrote 1270928 bytes (821711 compressed) at 0x00001000 in 19.4 seconds (effective 525.2 kbit/s)...
Hash of data verified.

But your log shown:
Flash will be erased from 0x00001000 to 0x00033fff...
Warning: Image file at 0x1000 doesn't look like an image file, so not changing any flash settings.
Compressed 208112 bytes to 36251...
Writing at 0x00001000... (33 %)
Writing at 0x00017750... (66 %)
Writing at 0x0002fa4b... (100 %)
Wrote 208112 bytes (36251 compressed) at 0x00001000 in 3.9 seconds (effective 427.0 kbit/s)...

That is why I thing the firmware you are trying to flash was wrong.  Maybe the firmware somehow get modified by renaming, copying in the Windows system. I suggest you try to down load the firmware and check it size. I do not have a Windows PC so I can not make the test.

deus-est-machina, 2023-02-28:
O_o

It definitely has something to do with the `Save link...` from browser.

I've never seen that hehe always a first for everything. After cloning the repo it looks much better. (I was thinking the 200kB weird... but it didn't even click for me until you said it)

[image]

I'll flash it again when I'm at the office in the morning and let you know.

deus-est-machina, 2023-03-02:
<3 you are a life saver! Thank you. After flashing the properly downloaded firmware, it works like a charm.

closed 2023-03-02 by deus-est-machina