shariltumin/esp32-cam-micropython-2022 #16: Parameters in camera.init

e16384, 2023-03-06:
Dear Mr. Sharil,
I've installed your firm **v1.18-610-gcf7d962cf-kaki5**  (on ESP32Cam), when I try to do this:
```
import camera
camera.init(0, xclk_freq=20000000) 
```

I have the error: _TypeError: function doesn’t take keyword arguments_

Then, how can I change the clock frequency?

Another question, when _camera.init()_ fails, I have tried to do _camera.deinit()_ several times waiting for a second between one attempt and another, but _init()_ still fails.
The only solution I've found is to do a reset.
Is there any other solution?

Best regards.

shariltumin, 2023-05-21:
Please have a look at [firmwares-20230521](https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/firmwares-20230521)

The new firmware supports
```
>>> import camera
>>> camera.
aecvalue        aelevels        agcgain         brightness
capture         conf            contrast        deinit
flip            framesize       init            mirror
pixformat       quality         saturation      speffect
whitebalance
```
What you want is camera.conf method.

e16384, 2023-05-22:
Thank you very much, I will try it out.

thecatinbed, 2023-05-26:
> 
I do enter "camera." after "import camera", but the system prompts syntax error. The firmware version is the latest, I can't figure out it.

shariltumin, 2023-05-26:
The firmware does not include help(). As a result, you will not be able to list all methods from the camera module. In the REPL example shown in the README, "camera." means "camera.\<TAB\>". This is the autocomplete feature of the MicroPython REPL. When a tab character is placed after a module name and a period, all possible attributes following the module are listed. The \<TAB\> key is usually located in the upper left corner, just below the digit keys, and has an arrow with a bar symbol.