shariltumin/esp32-cam-micropython-2022 #18: I need booth camera and bluetooth classic

maomiaomi, 2023-05-19:
Hello, may I kindly ask why the code needs to be deleted? Is there a possibility of releasing the code in the future? I am interested in using both a camera and Bluetooth classic on my car, and I wanted to know if you have a micropython firmware that supports both functions simultaneously? Alternatively, if it's convenient for you, could you share the camera code?
Thank you very much for your assistance.

shariltumin, 2023-05-21:
Please have a look at [firmwares-20230521](https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/firmwares-20230521)

andim27, 2023-05-22:
Great! Thank's a lot!

shariltumin, 2023-06-05:
A firmware with both camera and Bluetooth Classic can be found [here](https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/firmwares-20230521/BTC).