shariltumin/esp32-cam-micropython-2022 #20: How to record video

shaguahehehe, 2023-05-28:
How to record video

shariltumin, 2023-05-29:
The image is in jpg format. We have no support the output formats avi or mp4. You can capture a jpg image in a loop and save it to a file. See the following example. 

```
import camera, machine, uos, gc
from time import sleep

# init camera
camera.init()

# set preffered camera setting
camera.framesize(10)     # frame size 800X600 (1.33 espect ratio)
camera.contrast(2)       # increase contrast
camera.speffect(2)       # jpeg grayscale
img=camera.capture()     # throw away 1.st frame

# jpg file storage
sd=machine.SDCard(slot=1)
uos.mount(sd, "/sd")

for i in range(100): # 100 frames video
    img=camera.capture()
    if len(img) > 0:
       print('Frame:', i)
       fn = '/sd/pic-%03d.jpg'%i
       f = open(fn, 'wb')
       f.write(img)
       f.close()
       del f
    del img
#    sleep(60)
    gc.collect()

uos.listdir("/sd")
uos.umount("/sd")

```

Now that you have a large number of jpg files, you can convert them to videos. You can use ffmpeg on a Linux computer. 

```
$ ffmpeg -framerate 10 -i /media/sharil/0E0B-1721/pic-%03d.jpg video.mp4
```

or

```
$ ffmpeg -framerate 10 -i /media/sharil/0E0B-1721/pic-%03d.jpg video.avi
```

A low-quality micro-SD card is one item that can cause problems and frustration. If your SD card fails, your script may hang during the writing process. Don't go for the cheapest card you can find on the internet, because that will only cause problems.

shaguahehehe, 2023-05-30:
I tried to use Bluetooth to send a instruction to control the camera, but once the Bluetooth emitted the video, it was not closed when the Bluetooth sending the stop recording video was not closed. How can I solve it?

shariltumin, 2023-05-30:
I'm not sure how to respond to your question. I'm not sure what you're trying to accomplish. Some specifics and script examples may be useful.

shaguahehehe, 2023-05-31:
I tried to use Bluetooth to send a instruction to control the camera, but once the Bluetooth emitted the video, it was not closed when the Bluetooth sending the stop recording video was not closed. How can I solve it?

ble = ESP32_BLE("ESP32BLE")
        count = 1
        flag = False
        video_path = ''
        video_file = None
        
        while True:
            if BLE_MSG == 'open':        
                video_path = '/sd/video' + str(count) + '.avi'
                video_file = open(video_path, 'wb')
                print("开始录制")
                flag = True
                while flag:
                    for i in range(100):
                        img = camera.capture()
                        video_file.write(img)                    
                    if BLE_MSG == 'close':
                        flag = False
                        print("关闭录制")
                        BLE_MSG = ""          
                        video_file.close()
                        camera.deinit()
                        print('Video saved to', video_path)
                count += 1          
              
            sleep_ms(10)

shariltumin, 2023-05-31:
I'm not familiar with the specifics of the ESP32_BLE module. To begin, I'm not sure where in your script the BLE_MSG variable gets its value; let's assume it was set by BLE_MSG = ble.get_message(). Let's simplify your script and see if you ever get BLE_MSG == 'close'.

```
ble = ESP32_BLE("ESP32BLE")
while 1:
  BLE_MSG = ble.get_message()
  if BLE_MSG == 'open':
     print('OPEN')
  elif BLE_MSG == 'close':
     print('CLOSE')
  else:
     if BLE_MSG:
        print('Unknown cmd:', BLE_MSG)
  sleep_ms(10)
```

Perhaps ESP32_BLE() is not functioning as expected. You should seek assistance from those who created the module.

shaguahehehe, 2023-05-31:
Bluetooth can receive the Open instructions that will enter the circular video, unless the withdrawal of the cycle can receive the next information sent by Bluetooth, is there any way to receive Bluetooth information asynchronously

shariltumin, 2023-05-31:
What output you get when you run the simple script above? Try to send different strings from the BLE client.

shaguahehehe, 2023-05-31:
Simple examples are completely possible, but what I want to solve is how to allow Bluetooth to receive data asynchronously

shaguahehehe, 2023-05-31:
I want to see eSP32_ble.py

shaguahehehe, 2023-05-31:
Bluetooth receives the information to turn on the camera command, will continue to loop recording, if not exit the loop, then Bluetooth will not receive the next command

shariltumin, 2023-05-31:
The esp32_ble.py file can be found [here](https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/firmwares-20230521/BLE/example)

shaguahehehe, 2023-06-01:
The same problem I want to solve with your script is that when Esp32 receives a 'start' it will run and can't always send a 'stop' to stop it

shariltumin, 2023-06-01:
I'm running out of ideas on what to do to fix this problem. It could be a hardware problem. It could be a problem with the SD card. Please run the test script  [test_sdcard.py](https://github.com/shariltumin/esp32-cam-micropython-2022/blob/main/firmwares-20230521/SDcard/test_sdcard.py) to verify that the SD card is OK.