shariltumin/esp32-cam-micropython-2022 #21: #how to stream video from esp32 cam to tkinter python with link http://192.168.0.102:80/webcam ?

ktap23k, 2023-05-30:
i tried using cv2 but it can't read the format of this url http://192.168.0.102:80/webcam, i also tried curl but it doesn't work, although on chrome browser it works fine. Do you have any solution for my problem? Thank you very much!

shariltumin, 2023-05-30:
Start the webcam webserver

```
>>> import webcam
...
...
PWD: UajaoH5S
Start server 80
Try - http://192.168.4.44/login/UajaoH5S
```

Now, on a linux computer, perform the following:

```
$ cd /tmp

$ curl http://192.168.4.44/login/UajaoH5S
OK!

$ curl http://192.168.4.44/live --output live.mp4
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 17.4M    0 17.4M    0     0   196k      0 --:--:--  0:01:30 --:--:--  214k^C
```

Note: \<ctrl\>+c to kill curl

You will now have a file called live.mp4, which contains a video file that you can watch on celluloid.

```
$ celluloid live.mp4
```

This is probably the simplest way to capture video from an esp32-cam.
Take note of the URL! It's http://xx.xx.xx.xx/live, not../webcam, as you stated.

You can skip the login step if you disable authentication.

I've never used openCV, so I can't help you there.

ktap23k, 2023-05-30:
oh, looks like the problem was with the authentication when using with opencv or something similar not web (i use it on mac maybe there is some difference), now it works for me, thanks he is very much about this module!