shariltumin/esp32-cam-micropython-2022 #22: LILYGO  T-Camera S3 ESP32-S3 ESP32?

jouellnyc, 2023-06-02:
Hi. Firstly this repo is amazing. I really appreciate what you've done. I've revived old ESP32-CAMS to good effect.

Question:

I see in https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/firmwares-20230521

you mention: 

"...working on firmwares for esp32-wrover, esp32-s2, esp32-s3"

I just picked up  this board to use in a project: https://www.tindie.com/products/lilygo/lilygo-t-camera-s3-esp32-s3-esp32-cam-2-million/

But, if you need a tester when the newer FW comes out, I am happy to donate my time to do so.

jrmfox, 2023-06-05:
I agree, I like this repo very much. Thank you for the tip for the lilygo board, it looks interesting and I am considering ordering some. Did you successfully install the firmware on it? I am working on a Micropython project using a AI-Thinker ESP32-CAM and my objective is to send Images via MQTT to a broker and find out what is possible.

jouellnyc, 2023-06-11:
Howdy @jimrafox  

I've only dabbled, but I recall getting the  generic firmware (w/o camera) to load on the new board:

```
MicroPython v1.19.1-915-g2bcd88d55 on 2023-03-02; ESP32S3 module (spiram octal) with ESP32S3
Type "help()" for more information.

```

jimrafox, 2023-06-12:
Ok, thanks for sharing. I will consider using these.

shariltumin, 2023-06-17:
I'm waiting for my camera board from China to arrive. When it arrives, I'll try to write MicroPython firmware for it. Perhaps by next month, we'll have a working firmware.

jouellnyc, 2023-06-26:
Cool. If I can help you in in mean time, happy to try

shariltumin, 2023-07-17:
Ok, you can find the firmware [here](https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/firmwares-20230717/ESP32-S3/LILYGO-T-Camera-OV2640).

jouellnyc, 2023-08-13:
Thanks for that. I tried but am having some difficulty actually flashing my esp32. Not sure if it's Thonny or otherwise. I can get a REPL but can't upgrade the firmware (tried holding down both buttons to get it into boot loading mode). It could be my PC...Will update here if I get it working...

jan-reitsma, 2023-08-29:
> Ok, you can find the firmware [here](https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/firmwares-20230717/ESP32-S3/LILYGO-T-Camera-OV2640).

Great! Thank you!