shariltumin/esp32-cam-micropython-2022 #23: No time.time() function in firmwares-20230521

ljweko, 2023-06-07:
I've flashed WiFi + TLS firmware 20230521 into ESP32 CAM, but can not use time.time() function:
```
>> import time
>>> time.time()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'time'
>>> dir(time)
['__class__', '__name__', '__dict__', 'sleep', 'sleep_ms', 'sleep_us', 'ticks_add', 'ticks_cpu', 'ticks_diff', 'ticks_ms', 'ticks_us']
>>>
```
I tried to recompile binaries by myself, but couldn't find any recent source code.

Thanks.

shariltumin, 2023-06-07:
Thanks for the report. There were errors in my mpconfigport.h file. As an experienced custom custom firmware builder, you know how much detail you have to pay attention to every every time a new version of MicroPython is released. There will be a lot of changes when MicroPython goes to ESP-IDF 5. We will see how difficult it will be when the time comes.

Now the time module is fixed.

```
MicroPython v1.20.0-162-gd080d427e-kaki5 on 2023-06-07; ESP32-CAM OV2640 w/SSL (KAKI5) with ESP32
>>> import time
>>> time.
gmtime          localtime       mktime          sleep
sleep_ms        sleep_us        ticks_add       ticks_cpu
ticks_diff      ticks_ms        ticks_us        time
time_ns
```

The new firmware can be downloaded [here](https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/firmwares-20230607/WIFI%2BTLS).

ljweko, 2023-06-08:
Yes, time module is now complete. Thanks for fast reply and good luck with ESP-IDF 5.

Now real test can begin.

closed 2023-06-08 by ljweko