shariltumin/esp32-cam-micropython-2022 #24: REPL not accessible anymore

jimrafox, 2023-06-07:
Hi. I am using firmware 20230521 WIFI-TLS. I have uploaded a Micropython script that runs in an endless loop, taking images, comparing them and sending a MQTT message when there are significant changes.
It works fine! However, I am not able to abort the program and return to REPL, edit the file or upload another one. 
In Thonny, pressing "Stop" makes the shell hang, no output, no prompt. In a terminal emulator, no reaction a to break command either. 

However, I see this error, when starting the ESP32-CAM: 
[image]

It might have to do with this problem. Any ideas?

jimrafox, 2023-06-08:
Thank you for your answer. I use the default config values from your `config.py` file now. And I put my loop in a seperate file. I found out that, in my case, the error `cam_hal: EV-EOF-OVF` occurs at the moment when my script writes the image buffer to a file. Now I set `XCLK_MHZ:10` in the config.py file. The error still appears when writing to a file. I will have to look into it deeper, it might have to do with access to PSRAM or RAM.

shariltumin, 2023-06-08:
Your sensor could be faulty. If you have another OV2640, please try it. I've had some experience with cam_hal: EV-VSYNC-OVF error. Using a slower XCLK value will usually reduce the occurrence of errors. You can also try removing and reconnecting the sensor to the connector.

jimrafox, 2023-06-09:
Solution: I added some (long) sleep times between opening the file and writing into it. No more errors! XCLK is at 12 Mhz, unchanged. Will find out which sleep times are possible.
 ```
fs = open(file_name, "wb")
                sleep_ms(500)
                fs.write(img)
                sleep_ms(500)
                fs.close()
```

jimrafox, 2023-06-09:
Writing to flash, I need to sleep at least around 200 ms before writing to the file stream, at lower sleep intervals, the EV-EOF-OVF errors appear again. I have an SDCard and will try out what happens there.

closed 2023-06-11 by jimrafox