shariltumin/esp32-cam-micropython-2022 #26: camera module source

SilverLogix, 2023-06-11:
where is it at I want to compile it into my own custom firmware

shariltumin, 2023-06-11:
Building and maintaining custom firmware is a complicated undertaking. You need to track your board definitions like manifest.py, mpconfigport.h, sdkconfig.board. You also need to track changes to mpconfigport.h, main/CMakeLists.txt, main.c, idf components, your special module in the main ports.

esp32-cam has limited RAM resources, so you have to be careful what you include in your in your firmware. Some standard modules in MicroPython are tightly coupled, such as thread and thread and Bluetooth.

I choose not to share the source because it will cause a lot of problems. Have you ever tried to build custom firmware for a custom board? If you have, you know what I am talking about.

By the way, I have a book project on DIY cameras for esp32-type boards for those interested like you. I have to put that on hold for a moment, since MicroPython for esp32 ports is now moving to esp-idf-5.

You cannot find the source in my repository, maybe you can have a look at the [lemariva](https://github.com/lemariva/micropython-camera-driver) repository.

SilverLogix, 2023-06-12:
I will look into lemariva then or spend 3 days converting the original c source lol

bole5, 2023-08-19:
@shariltumin It is great that you are providing firmware ready to use for less experienced users. This issue was opened by someone who wanted to compile their own firmware. I personally do not see why providing source code would be problematic. If anything, it would encourage github community to contribute and improve camera firmware.

SilverLogix, 2023-08-19:
And considering I test, contribute, and sponsor micropython, I might know a little bit of what I'm doing, haha

shariltumin, 2023-08-20:
The lemariva repository mentioned above is a good place to start if you want to build your own firmware to support the OV2460. The 'modcamera.c' is nicely written and you can build on it.

You might also want to have a look at [esp-camera](https://github.com/espressif/esp32-camera)

If you can manage to build a working firmware from them (I'm sure you can!), you can have your own repository.

I would like to remind you that ESP-IDF, esp-camera component, MicroPython build process all change rapidly. Maintaining your repository can be a challenge. Things can quickly turn from fun to pain.

You can also try the (MP Discussion Forum)[https://github.com/orgs/micropython/discussions], there are lots of clever people there if you need help.

Sorry, I have a plan with my source code and at the moment I do not intend to make it public.