shariltumin/esp32-cam-micropython-2022 #27: Parameters in camera.pixformat(x)

xhgeya, 2023-07-03:
Dear Mr. Sharil,
Thanks for the firms and your hard works! 
I've installed your firm v1.20.0-162-gd080d427e-kaki5 on 2023-06-07; ESP32-CAM OV2640 w/SSL (KAKI5) with ESP32 for my ESP32-CAM, and these steps are:

import camera, machine
# init camera
cam = camera.init()
if cam:
   print("Camera ready?: ", cam)
else:
   camera.deinit()
   machine.reset()
# set preffered camera setting
camera.framesize(1)     # 1:96x96
camera.pixformat(0)     # 0:JPEG
img=camera.capture()     
print(len(img))
camera.deinit()

Thonny shows the right result if camera.pixformat(0) or without it.

***********
Camera ready?:  True   
1784
***********

Once I try to do these: 
camera.pixformat(1)  #  1:Grayscale (2bytes/pixel)
or
camera.pixformat(2)  #  2:RGB565

I have got the error:  TypeError: object of type 'bool' has no len()

Read from help.py, it shall be ok since pixformat has three parameters(0:JPEG, 1:Grayscale (2bytes/pixel), 2:RGB565)
What shall I do? Thanks！
Good day!
George

shariltumin, 2023-07-03:
Thanks for reporting this. It appears that you have caught a bug that was reported [here] (https://github.com/espressif/esp32-camera/issues/436).

The default image format is JPEG. This only happens for 96x96:JPEG. For earlier firmwares, this bug will persist, i.e. the lowest framesize for jpeg will then be 2:160x120 or 5:240x240 for a square image.

The help.py is no longer correct, the note on pixformat is wrong.

MicroPython v1.20.0-206 was the last version built with esp-idf-4. All MicroPythons after that will only build with esp-idf-5.

I am working on new firmwares. These firmwares will use esp-idf-5.x.x, and with some minor fixes in the camare module, the 96x96:JPEG bug seems to be fixed.

```
>>> 
MPY: soft reboot
MicroPython v1.20.0-242-g6a9db521e on 2023-07-03; ESP32 AIT OV2640 (KAKI5) with ESP32
>>> import camera
>>> camera.init()
True
>>> camera.framesize(1)   # 96x96
>>> camera.pixformat(4)   # Grayscale
>>> img=camera.capture()
>>> len(img)
1043
>>> camera.pixformat(5)   # jpeg
>>> img=camera.capture()
>>> len(img)
1027
>>> img=camera.capture()
>>> print(len(img))
1079
```
The help.py needs to be fixed. As soon as I finish testing, I will upload the latest firmwares to the github.

If you need to use 96x96:JPEG, you will have to wait a little bit longer.

xhgeya, 2023-07-07:
Thanks a lot! 
👍

closed 2023-07-07 by xhgeya