shariltumin/esp32-cam-micropython-2022 #28: Firmware for lilygo camera v1.6.2

adrianlzt, 2023-07-07:
I have created a repo to build the micropyhton firmware with camera for the lilygo camera v1.6.2

It contains the compiled firmware and the instructions (in Dockerfile form): https://github.com/adrianlzt/micropython-esp32-cam-docker

Hope it could help.

shariltumin, 2023-07-07:
Thanks, nicely done. I recommend your [micropython-esp32-cam-docker](https://github.com/adrianlzt/micropython-esp32-cam-docker) and [micropython-camera-driver](https://github.com/adrianlzt/micropython-camera-driver) to those who want to make their own firmware. Following your example, it would not be difficult to make firmware for other boards.

I follow different messy way since I did more customization than you did. Yours is much cleaner and better.

I see that you are now building firmware with esp-idf-502. What is your experience? I am having problems with machine.timer(), I am getting Guru Meditation Error: Core 1 panic'ed (Interrupt wdt timeout on CPU1), see [#11948](https://github.com/orgs/micropython/discussions/11948).

I am tempted to freeze my firmwares at esp-idf-445 and do the migration at a later point in time, when some of the issues have been fixed.

Again, well done.

adrianlzt, 2023-07-08:
I haven't tested much still, but yes, timer is broken also in mine build. I have answered in #11948