shariltumin/esp32-cam-micropython-2022 #32: no module named 'onewire'

nirki123, 2023-07-30:
Hi, I'm trying to use the onewire and ds18x20 modules and I'm using the firmware version from the main directory.
I have read that they are supposed to be built-in in the firmware.
Is the library not supported by this firmware?
Should I upgrade to a newer firmware version?

Thanks, Nir.

shariltumin, 2023-07-30:
I need to decide what to include in the firmware that supports the OV2640 camera module. Some 'standard libraries' were not included in the hope to save some RAM. Modules 'onewire' and 'ds18x20' were not included. 

How you use the onewire and ds18x20 with the camera in your system? Do you need to read a temperature reading to trigger a photo capture? I could probably squeeze those two in if they are related to image capture in your system.

These are the contents of manifest.py:

```
#freeze("$(PORT_DIR)/modules")
freeze("$(PORT_DIR)/modules", ("_boot.py", "flashbdev.py", "inisetup.py"))
include("$(MPY_DIR)/extmod/uasyncio")

# Require some micropython-lib modules.
#require("dht")
#require("ds18x20")
#require("neopixel")
require("ntptime")
#require("onewire")
require("umqtt.robust")
require("umqtt.simple")
require("upysh")
require("urequests")
#require("webrepl")
```

nirki123, 2023-08-02:
I don't use the camera. I just only have the esp32-cam currently. I'll just wait for the regular esp32 I ordered.
Thanks.

bole5, 2023-09-07:
I would rather include webrepl if this is not too much. Being able to quickly check remotely what is happening with espcam would be great.