shariltumin/esp32-cam-micropython-2022 #33: Package manager is not included in the builds

bole5, 2023-08-19:
Hello. I am using latest build with SSL. In this build there is no possibility to install additional packages using pip/mip.
Can you include package manager in your builds?

```
MicroPython v1.20.0-206-g33b403dfb-kaki5 on 2023-07-11; ESP32 CAMERA w/SSL (KAKI5) with ESP32
>>> import mip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'mip'
>>> import upip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'upip'
>>> import pip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'pip'
```

shariltumin, 2023-08-20:
I will not be compiling new firmwares at the moment. There are a few bugs I need to fix first.

For now, you can install mip by following these instructions:

The mip package is located in micropython/lib/micropython-lib/micropython/mip/mip/\_\_init\_\_.py, you need to copy this file to a file called mip.py.

cp from micropython/lib/micropython-lib/micropython/mip/mip/\_\_init\_\_.py mip.py

Upload mip.py to your esp32-cam board

```
>>> # connect to your wifi
>>> import time, network, socket
>>> ap = network.WLAN(network.AP_IF); ap.active(False)
>>> nic = network.WLAN(network.STA_IF); nic.active(True)
>>> nic.connect("ssid", "passwd")
>>> print(nic.ifconfig())
>>> import mip
>>> mip.install('mip')
Installing mip (latest) from https://micropython.org/pi/v2 to /lib
Copying: /lib/requests/__init__.mpy
Copying: /lib/mip/__init__.mpy
Done
>>> os.listdir('/lib')
['mip', 'requests']
```

bole5, 2023-08-21:
Thank you for the workaround.

I got an error that `requests` was missing which was resolved by downloading below file and renaming it to requests.py:
`https://github.com/micropython/micropython-lib/blob/master/python-ecosys/requests/requests/__init__.py`