shariltumin/esp32-cam-micropython-2022 #34: FREENOVE OV2640 installation problem

CRichard78, 2023-09-27:
Hello.
I am new in micropython, I have a FREENOVE ESP32-S3 WROOM N8R8 board with OV2640 camera.
I want to try micropython with this board. 
First attempt is to flash the board with the provided firmware and every thing is OK. THONNY is working fine.
Second attempt is to flash with the firmware in the folder "firmwares-20230717/ESP32-S3/Freenove-OV2640/Wifi or Wifi+BLE/firmware.bin. The result is the same with or without BLE.
I follow the installation procedure with esptool with the command. To test the command I used the same command for flashing the provided firmware with success.
python esptool/esptool.py --port COM38 --baud 460800 --before default_reset --after hard_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 firmware.bin

Every thing is working fine for flashing the board but THONNY says that device is busy 
Thonny shell window displays 
*******************
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0xf58
load:0x403c9700,len:0xc3c
load:0x403cc700,len:0x2e48
entry 0x403c98f0

Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - reset the device and try again;
  - check connection properties;
  - make sure the device has suitable MicroPython / CircuitPython / firmware;
  - make sure the device is not in bootloader mode.
***************
Please could you help me?
Best regards.
Claude.

shariltumin, 2023-09-27:
Can you try connecting your board using a serial terminal program like minicom ( Tera Term, CoolTerm, etc)? I assumed you are on Windows.

CRichard78, 2023-09-27:
Thanks for your answer.
With PUTTY display is 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0xf58
load:0x403c9700,len:0xc3c
load:0x403cc700,len:0x2e48
entry 0x403c98f0

shariltumin, 2023-09-27:
And the terminal just hang?  Try <enter> or Ctrl-C or Ctrl-D

CRichard78, 2023-09-27:
No just hang. Ctrl-C or Ctrl-D do nothing.

CRichard78, 2023-09-27:
I think that I found.

There are 2 USB ports on the board. 
If I use the port below the BOOT button I can flash the ESP32 but can't use micropython.
If I use the port below the RST button, I can use micropython.
I will try to flash the ESP32 with this port.
Thank you for your help.
Best ragards.
Claude.