shariltumin/esp32-cam-micropython-2022 #36: Failed to initialize camera in Thonny

gem7008, 2023-09-30:
Hi. I have flashed the following firmware: `firmwares-20230717/ESP32/AI-Thinker-OV2640/WiFi-SSL` into my AI-thinker ESP32 Cam.

The firmware works as intended, however, I have issues running codes from the Thonny IDE (via the shell) as when I run code through Thonny's shell, I am not able to initialize the camera (`camera.init()` will always return False).... however, running the code on the Esp32cam (by rebooting it and have the code in `main.py`) works just fine.

I would like to be able to run it through the Thonny's shell in order to be able to easily debug and see exception. Do you know by any chance what could be the issue and how I could mitigate it?

shariltumin, 2023-10-01:
As you already knew, "main.py" is run at boot time. If the file contains ```camare.init()```, you will get an error if you try to run it again in the REPL.

You can either delete the "main.py" stored in the flash or store a new empty "main.py" in the flash.

Or you can try ```camare.deinit()``` before ```camare.init()``` while you are in the REPL.

gem7008, 2023-10-01:
Thanks for the help. I can confirm that deleting the `main.py` stored in the flash worked. (didn't try other solutions, but I believe that will also work)..

closed 2023-10-01 by gem7008