shariltumin/esp32-cam-micropython-2022 #43: Help with using the SD card on the Freenove ESP32-S3 Board

ArevomEisvolk, 2024-01-04:
I would like to build a wildlife camera with your firmware, not very complex but quite nice, in my opinion. Now I am facing the issue that something seems to be wrong with the SD card library 'SD MMC peripheral on ESP32-S3 #10598'. I had to recompile the firmware with the fix shown here. Unfortunately, this is not yet available. Would you please recompile the firmware with the fix?

**SDcard Pin** - From the C Tutorial PDF
An SDcard slot is integrated on the back of the ESP32-S3-WROOM board. We can use GPIO38-GPIO40 of
ESP32-S3-WROOM to drive SD card.
The SDcard of ESP32-S3-WROOM uses SDMMC, a 1-bit bus driving method, which has been integrated in
the Arduino IDE, and we can call the "SD_MMC.h" library to drive it. For details, see the SDcard chapter in this
tutorial.

**SDcard Pin** - From the Python Tutorial PDF
An SD card slot is integrated on the back of the ESP32-S3 WROOM board. We can use GPIO38-GPIO40 of
ESP32-S3 WROOM to drive the SD card.

https://github.com/orgs/micropython/discussions/10598

shariltumin, 2024-01-04:
Hello,

Sorry about that. Give me a few days to work on a new firmware with some bug fixes, and which will also include the onboard SD card drive.

Thanks for the tips, by the way.

ArevomEisvolk, 2024-01-17:
Hallo, are there any updates, are you making progress?

shariltumin, 2024-01-21:
Please find the new firmware [here](https://github.com/shariltumin/esp32-cam-micropython-2022/tree/main/X23/esp32s3-freenove)

ArevomEisvolk, 2024-01-21:
Many thanks, it works **perfectly**.

closed 2024-01-21 by ArevomEisvolk