shariltumin/esp32-cam-micropython-2022 #47: machine.reset() sometimes fails

davefes, 2024-04-13:
Latest image on a ESP32 Generic.

When doing a machine.reset() sometimes get the following:

E (3070) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (3070) camera: Camera Init Failed
E (3070) camera: Camera deinit Failed
E (4130) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
E (4130) camera: Camera Init Failed
E (4130) camera: Quality Failed
E (4130) camera: Special Effect Failed

A power-on-reset clears this problem.  Any suggestions?

Thanks

shariltumin, 2024-04-13:
Have you try ```camera.deinit()``` before calling ```machine.reset()``` ?

davefes, 2024-04-13:
I will try that.  It is a random event.

I start every program with:
```py
# during development
if machine.reset_cause() == machine.SOFT_RESET:
    print('you got a SOFT_RESET, so do a machine.reset()')
    time.sleep(1)
    machine.reset()
```
and it fails here before any camera init happens.

davefes, 2024-04-14:
Putting `camera_deinit()`gives a `E (1070) camera: Camera deinit Failed` which is not surprising as the camera has not been initialised.  My original error appeared to go away.

If I `import camera` after my `SOFT_RESET` test everything works properly.

Thank you.

closed 2024-04-14 by davefes