shariltumin/esp32-cam-micropython-2022 #48: camera API, where can I find it?

davefes, 2024-04-17:
In your older commits I found:
```
camera.pixformat(0)   # 0:JPEG, 1:Grayscale (2bytes/pixel), 2:RGB565
camera.framesize(10)  # 1:96x96, 2:160x120, 3:176x144, 4:240x176, 5:240x240
                      # 6:320x240, 7:400x296, 8:480x320, 9:640x480, 10:800x600
                      # 11:1024x768, 12:1280x720, 13:1280x1024, 14:1600x1200
                      # 15:1920x1080, 16:720x1280, 17:864x1536, 18:2048x1536
camera.quality(5)  # [0,63] lower number means higher quality
camera.contrast(0)  # [-2,2] higher number higher contrast
camera.saturation(0)  # [-2,2] higher number higher saturation. -2 grayscale
camera.brightness(0)  # [-2,2] higher number higher brightness. 2 brightest
camera.speffect(2)  # 0:,no effect 1:negative, 2:black and white, 3:reddish,
                        # 4:greenish, 5:blue, 6:retro
camera.whitebalance(0)  # 0:default, 1:sunny, 2:cloudy, 3:office, 4:home
camera.aelevels(0)  # [-2,2] AE Level: Automatic exposure
camera.aecvalue(0)  # [0,1200] AEC Value: Automatic exposure control
camera.agcgain(0)   # [0,30] AGC Gain: Automatic Gain Control
```

but where in the ESP_IDF documentation can I find some explanation?
I seem to be getting some over-exposed pictures so maybe aelevel,
aecvalue and/or agcgain need tweaking from the default values.

Thank you,
Dave

shariltumin, 2024-04-17:
Those are camera module specific info not ESP_IDF. 

Please search this in google
OV2640 Camera Module Software Application Notes.pdf

You can ESP-IDF camera component source code here, https://github.com/espressif/esp32-camera

davefes, 2024-04-17:
OK, thanks.

shariltumin, 2024-04-18:
There is no proper manual concerning the camera component. I did guess works and experiments. 

The parameters to the calls can be found here, https://github.com/espressif/esp32-camera/blob/master/sensors/ov2640.c

Sorry, for not being helpful, but this is all I have.

davefes, 2024-04-18:
Actually, that does have some information about those values in it (down around line 500).  Yesterday I tried changing the saturation and ended up getting a colour picture!?!

More guess work required.

Thanks again.

closed 2024-04-18 by davefes