shariltumin/esp32-cam-micropython-2022 #61: GPIO13 PULL_DOWN doesn't seem tp work

davefes, 2025-01-07:
PULL_UP goes to 3.2V, PULL_DOWN is at 0.9V and with no parameter it sits at 0.9V

A generic ESP32 behaves as expected.  I am happy to use external pull down resistors but curious to know if this is to be expected.

Thanks.

shariltumin, 2025-01-07:
According to the schematic, GPIO13 was designed to be used as HS2DATA3 connected to MicroSD and permanently connected to 3V3 by 47K resistor.

You can get more information from https://randomnerdtutorials.com/esp32-cam-ai-thinker-pinout/

davefes, 2025-01-07:
Thank you.  The link does say:

> If you’re not using the microSD card, you can use these pins as regular inputs/outputs. 

Because the schematic didn't have actual lines on it, it was not obvious that these points were connected together.

Also, it appears that if you don't specify PULL_DOWN or PULL_UP then you get a PULL_DOWN by default.

I appreciate the clarification.

closed 2025-01-07 by davefes